stooged/PI-Pwn #152: question?

mihaiaka, 2024-06-01:
is it normal that these 3 steps go INSTANTLY? And not really line-by-line? I understand that the new C++ is faster, but the Windows tool or the Router tool still does them line by line.

[*] Heap grooming...0%
[*] Heap grooming...6%
[*] Heap grooming...12%
[*] Heap grooming...18%
[*] Heap grooming...25%
[*] Heap grooming...31%
[*] Heap grooming...37%
[*] Heap grooming...43%
[*] Heap grooming...50%
[*] Heap grooming...56%
[*] Heap grooming...62%
[*] Heap grooming...68%
[*] Heap grooming...75%
[*] Heap grooming...81%
[*] Heap grooming...87%
[*] Heap grooming...93%
[+] Heap grooming...done

[*] Pinning to CPU 0...00%
[*] Pinning to CPU 0...06%
[*] Pinning to CPU 0...12%
[*] Pinning to CPU 0...18%
[*] Pinning to CPU 0...25%
[*] Pinning to CPU 0...31%
[*] Pinning to CPU 0...37%
[*] Pinning to CPU 0...43%
[*] Pinning to CPU 0...50%
[*] Pinning to CPU 0...56%
[*] Pinning to CPU 0...62%
[*] Pinning to CPU 0...68%
[*] Pinning to CPU 0...75%
[*] Pinning to CPU 0...81%
[*] Pinning to CPU 0...87%
[*] Pinning to CPU 0...93%
[+] Pinning to CPU 0...done

[*] Scanning for corrupted object... 0xf00
[*] Scanning for corrupted object... 0xe00
[*] Scanning for corrupted object... 0xd00
[*] Scanning for corrupted object... 0xc00
[*] Scanning for corrupted object... 0xb00
[*] Scanning for corrupted object... 0xa00
[*] Scanning for corrupted object... 0x900
[*] Scanning for corrupted object... 0x800
[*] Scanning for corrupted object... 0x700
[*] Scanning for corrupted object... 0x600
[*] Scanning for corrupted object... 0x500
[*] Scanning for corrupted object... 0x400
[*] Scanning for corrupted object... 0x300
[*] Scanning for corrupted object... 0x200
[*] Scanning for corrupted object... 0x100
[*] Scanning for corrupted object... 0x000
[-] Scanning for corrupted object...failed.

PI-Pwn has the worst success rate for me. It takes 5-10-15 tries for me and if I get successfully jailbreak, but it doesn't show the GoldHEN message and freezes/reboots/black screen in a few seconds.
The Windows PPPwd GUI (moded warfare) does the trick in 2-3 tires and no problems.
The Router option (xfangfang - arm-linux-musleabi(mpcorenovfp)) did the trick the first go.

Last week (prior some updates) it worked better/more stable.

Thank you,
M.

TraPstonereal, 2024-06-01:
i´m trying right now 6 hours to get it work but it sits on deafiting KASLR
[image]

TraPstonereal, 2024-06-01:
and my ps4 hat blackscreen and all this thing

stooged, 2024-06-01:
More lines on the debug output isnt going to change the success rate.
If the windows one is quicker then its probably best to use that.

The pppwn exploit is what it is and the c++ port will probably always run better on wondows.

closed 2024-06-01 by stooged