stooged/PI-Pwn #160: Can that pwn tool used on esp32 with Wi-Fi ( without ethernet module) ?

marhau-dev, 2024-07-06:
[esp32 with wifi](https://www.aliexpress.us/item/3256806150650156.html?spm=a2g0o.productlist.main.1.6cc9ffEWffEWH0&algo_pvid=da740f52-2ec7-43dd-a445-c6c9295432b1&algo_exp_id=da740f52-2ec7-43dd-a445-c6c9295432b1-0&pdp_npi=4%40dis%21USD%217.69%210.99%21%21%2155.74%217.13%21%402103080b17202302045192771eeaf6%2112000036806447870%21sea%21US%216022491456%21AB&curPageLogUid=yib78ZtA5TZm&utparam-url=scene%3Asearch%7Cquery_from%3A)

Doyle4, 2024-07-06:
No

closed 2024-07-06 by marhau-dev

Doyle4, 2024-07-06:
@marhau-dev If you are wanting to do it cheap as possible, Luckfox is another way and simuliar to PI-PPPwn

https://www.aliexpress.com/w/wholesale-luckfox.html?spm=a2g0o.home.search.0

https://github.com/0x1iii1ii/PPPwn-Luckfox