stooged/PI-Pwn #164: Beta for Problem Model

Jorge07-07, 2024-07-09:
I have a console model that I can only unlock with the PSGO PPPwnGo / PPPwn-Lite aplications in the Beta version, Problem Model. Is it possible to have this Beta version on the Raspberry Pi?

stooged, 2024-07-09:
not sure what "C++ Beta CPP 0616" is supposed to be but i will assume you are talking about models of ps4 that were unable to pwn until the ipv6 address was modified.

by default pi-pwn uses the modified address for problematic consoles with an option to switch back to the original.

closed 2024-07-09 by stooged

Jorge07-07, 2024-07-10:
> not sure what "C++ Beta CPP 0616" is supposed to be but i will assume you are talking about models of ps4 that were unable to pwn until the ipv6 address was modified.
> 
> by default pi-pwn uses the modified address for problematic consoles with an option to switch back to the original.

Even with IPv6 I can only unlock my console by switching to the beta version that PSGO introduced in its application. Otherwise, it keeps trying to unlock without success.
https://github.com/PSGO/PPPwn-Lite/releases/tag/v5.6