stooged/PI-Pwn #165: network sharing issue

epiclulz666, 2024-07-14:
I'm using a Raspberry Pi 4 with Raspberry Pi Lite 64-bit installed. I've connected a USB-A to Gigabit adapter from my PS4 Pro to the Raspberry Pi 4, and I'm using Pi-Pwn for jailbreaking. The jailbreak interface is set to eth1, and I've successfully set up internet sharing. I can browse the web and perform internet activities on the PS4 using an Ethernet cable connected to a switch that's linked to the Pi's main Ethernet port.

However, my issue is that internal functions aren't working as expected. I've enabled the FTP server in GoldHEN, and from a PC on my network, I attempt to connect to the Pi's IP address on port 2121. It shows that the connection is established but doesn't display any directories. Although I can ping and SSH into the Pi, the remote package sender (which operates on port 12800, i also added the servers port of 8337 ) this also doesn't work. When I use the test button in the app, it reports that it can't find the PS4.

I tried adding routes into iptables based on suggestions I found, but this hasn't resolved the issue either. Do you have any suggestions on how to troubleshoot or fix this problem?

epiclulz666, 2024-07-14:
[image]

stooged, 2024-07-14:
https://github.com/stooged/PI-Pwn?tab=readme-ov-file#console-ftp--binload

closed 2024-07-14 by stooged