stooged/PI-Pwn #166: Request for the sources (and licenses!)

nn9dev, 2024-07-15:
Good afternoon,

This issue is being opened to request an up-to-date copy of the pppwn_cpp source code be published, as well as the license for it and the original PPPwn be added to the repository.

I truly do not want to sound patronizing or as though I am speaking down to you, please do not misunderstand. 

pppwn_cpp is licensed under GPLv3. To not have any confusion, I am simply informing that GPLv3 requires source code be published for any modified public distributions. PPPwn Python is just MIT, and only requires the license be included.

Additionally as a personal request of my own, your build process would be a good piece to publish. At present, I'm unsure how you go about generating your decimal byte arrays for payloads (and it doesn't make it very easily accessible).

Thank you for your time.

stooged, 2024-07-15:
https://github.com/stooged/PPPwn_cpp

nn9dev, 2024-07-15:
\> "an up-to-date copy of the pppwn_cpp code"

\> updated three weeks ago

\> looks at repo

\> binaries pushed 5 days ago

???

And... can I get an answer on anything else I brought up...?

stooged, 2024-07-15:
The GPL does not explicitly specify a time within which the source code must be provided, which means a "reasonable time" is allowed so weeks or months for an update would be reasonable.

only the copyright holder has standing so if xfangfang makes a request for any current code i have of pppwn_cpp i will provide it to him.

the original pppwn repo is here
https://github.com/stooged/PPPwn

the build process is listed in the pppwn_cpp readme
https://github.com/stooged/PPPwn_cpp?tab=readme-ov-file#development

nn9dev, 2024-07-15:
ok 👍