stooged/PI-Pwn #168: Automatic shutdown of Raspberry Pi when PS4 is turned off for a certain period of time

Mumuhaga, 2024-07-18:
Would it please be possible to implement an Automatic shutdown of the Raspberry Pi once the Raspberry Pi detect that the PS4 is turned off for a defined period of time (can be set to a fixed time e.g. 3 minutes or by the user via the web interface)
This would help us to do a safe shutdown of the Raspberry Pi without visiting the web interface.
Thank you very much for think about it. :-)

loddi-o, 2024-08-09:
you can do so with a bash script that keeps pinging the console, when ping is unsuccesfull for 3 minutes straight launch command shutown

DavidLeBonk, 2024-08-22:
You could also connect the Pi to one of the PlayStation USB ports, this way it'll turn off with the console.

closed 2024-08-27 by stooged