stooged/PI-Pwn #171: Performance statistics across models?

tester-testing-person, 2024-07-22:
I would like to know what the performance (how long it takes for an attempt) is across the raspberry pi models and what the failure rates are. You have listed the models you have tested in the readme but no stats for them, any chance you could add that information?

I would like to know what the difference between the Raspberry Pi 4 Model B and Raspberry Pi 5 are specifically.

Doyle4, 2024-07-23:
Raspberry Pi 4/5 had couple extra functions due to usb passthrough, other than that the 3b+ is almost identical.
I use the 3b+ powered via usb from the PS4 with a boot up to JB from 2-4mins, i poweron and leave it.

Better device can have a tad bit of extra speed, but not by much.

stooged, 2024-07-23:
Pi 4 and 5 are pretty much the same in speed and process the exploit much quicker than other pi variants.

As for benchmarking data across models i honestly never bothered with that and wouldnt because its just an exploit and if it takes 3 minutes or 5 it does not really matter because the end result is the same.

Its the same sort of thing with exfathax, people looking to shave 5 seconds off the exploit process for some strange reason.

closed 2024-07-23 by stooged