stooged/PI-Pwn #173: This doesn't work for me. :-(

antonycartwright, 2024-07-27:
I'm using RPi 2b. I've followed the usual guides.

I can do it successfully with my laptop, but not with the RPi. 

I'm using eth0 for the management console, and eth1 (USB) for the PS4. Regardless of which interface is connected to what, I get the same problem. It cycles as below, then after some time, it crashes the PS4, and it loses power (plugged into the PS4).

If I do not have the RPi connected, the PS4 doesn't crash, so it's definitely something to do with this.

Thoughts so far:
Authentication problem?
No hen.bin on USB?

[image]

Doyle4, 2024-07-30:
First time can take upto 5 mins.

waleedqq, 2024-08-07:
How did you get this output?

closed 2024-08-27 by stooged