stooged/PI-Pwn #180: goldhen.bin not found

lITronusIl, 2024-08-14:
hi my ps sys ppwned but after that it says goldhen not found. how to fix?

micaslarsen, 2024-08-15:
It's very important that you have the binary of GoldHEN on the root of a **exFAT** formatted USB drive.
The releases page can be found [here](https://github.com/GoldHEN/GoldHEN/releases)
The latest release is [GoldHEN_v2.4b17.3](https://github.com/GoldHEN/GoldHEN/releases/download/2.4b17.3/GoldHEN_v2.4b17.3.7z)
[image]

lITronusIl, 2024-08-15:
> It's very important that you have the binary of GoldHEN on the root of a **exFAT** formatted USB drive. The USB is only needed for the first PPPwn. The releases page can be found [here](https://github.com/GoldHEN/GoldHEN/releases) The latest release is [GoldHEN_v2.4b17.3](https://github.com/GoldHEN/GoldHEN/releases/download/2.4b17.3/GoldHEN_v2.4b17.3.7z) <img alt="Screenshot 2024-08-15 at 14 49 50" width="792" src="https://private-user-images.githubusercontent.com/89074001/358222929-7f123f9b-6152-4e4e-a677-96fee9444418.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM3NDkzOTAsIm5iZiI6MTcyMzc0OTA5MCwicGF0aCI6Ii84OTA3NDAwMS8zNTgyMjI5MjktN2YxMjNmOWItNjE1Mi00ZTRlLWE2NzctOTZmZWU5NDQ0NDE4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODE1VDE5MTEzMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYyMjllNjNlODhlZTY0MmM2NmMwZDA2YzczN2U2NTYwNzkxYzQzY2FkZTg3NDg3M2Q5OTAzY2EyN2QwYjUwYTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0._NhKXpQE3e8kyt4Gm1ler3rdF3lzlUVGZVeYNG1mBZk">

I solved the issue, i had use pi as usb drive enabled, that Interrupted it

closed 2024-08-27 by stooged