stooged/PI-Pwn #196: binloader failed to get ip

1bertox, 2024-10-29:
I don't know if this is the reason, but as soon as Pppwn is activated and loads Golden Hen, the message Binloader failed to get ip. aborting appears immediately.
I believe this is the reason why I can't access the web server and internet.
Is there any way to fix this?

[image]

Ps4 Pro - FW11.00

durag5, 2025-03-22:
What golden version?

ecchonight, 2025-03-22:
> I don't know if this is the reason, but as soon as Pppwn is activated and loads Golden Hen, the message Binloader failed to get ip. aborting appears immediately. I believe this is the reason why I can't access the web server and internet. Is there any way to fix this?
> 
> [image]
> 
> Ps4 Pro - FW11.00

Just disable PPPoE Patch on settings.