stooged/PI-Pwn #206: Console internet access not working

travman1974, 2024-12-30:
I have a raspberry pi 4 running Raspberry PI os (64-bit) bookworm. PS4 is connected to USB-Ethernet adapter on ETH1. ETH0 is connected to my router. PPPwn works with out issue, but shortly after a successful PI-PWN, a message will pop up saying LAN cable disconnected.  I am using the latest PI-PWN release, GoldHen 2.4b18, have enable console internet access checked. If I view network connection status IP address shows automatic.  I have also verified I am using the correct PPoE User Id and password.
Any help would be appreciated

durag5, 2025-01-05:
After the pwn it will immediately disconnect Ethernet for me as well.

lufeig, 2025-01-07:
Same happens for me. 

After pppwn it immediately disconnects eth0.

PS4 Pro Baikal 9.00
Raspberry Pi 4 4GB
Pi OS Lite 64bit
Raspberry Pi internet connection via wifi
Raspberry Pi console connection via built-in ethernet port eth0
Goldhen 2.4b18.2
Internet access enabled
FTP enabled
Samba enabled
Sleep patch enabled
Detect PS4 shutdown disabled

PPPOE user and password are correct on RPi and PS4: default "ppp".

Under system information, IP shows hyphen "-" after pppwn. Before pppwn it shows the correct IP 192.168.2.2.

ecchonight, 2025-01-07:
In my tests this a related GOLDHEN 2.1b18 error.
try with a old version and comment here.

lufeig, 2025-01-08:
> In my tests this a related GOLDHEN 2.4b18 error. try with a old version and comment here.

thank you very much for sharing your experience, you are absolutely correct! I tried it with goldhen 2.4b17.3 and the internet passthrough worked perfectly.

it seems that the payload allocation size has to be increased if we use 2.4b18 or 2.4b18.2.

mistersoup3, 2025-01-12:
I am running into the same issue -- I have tried switching from the packaged version of goldhen to 2.4b17.3  from https://github.com/GoldHEN/GoldHEN/releases , but am still having issues with the passthrough.

My setup is almost exactly the same as lufeig posted above, except on version 10.01, and on a PI 4 Model B [2gb]. The IP Address does still stay on the System information page after PWN is complete, but it does not seem to be connecting through the PI.

If I connect to the PI directly (via ssh) I am able to ping external websites (and also is definitely connecting to the LAN) so the issue doesn't seem to be there. I have re-ran the initialization as well to confirm that  I properly selected the 'Internet access enabled' option.

Is there anything else I should check? Would it be worth going back to even older versions of GoldHEN? Not totally sure where I could be going wrong / how to even troubleshoot where the problem lies. Thanks!

lufeig, 2025-01-15:
> I am running into the same issue -- I have tried switching from the packaged version of goldhen to 2.4b17.3 from https://github.com/GoldHEN/GoldHEN/releases , but am still having issues with the passthrough.
> 
> My setup is almost exactly the same as lufeig posted above, except on version 10.01, and on a PI 4 Model B [2gb]. The IP Address does still stay on the System information page after PWN is complete, but it does not seem to be connecting through the PI.
> 
> If I connect to the PI directly (via ssh) I am able to ping external websites (and also is definitely connecting to the LAN) so the issue doesn't seem to be there. I have re-ran the initialization as well to confirm that I properly selected the 'Internet access enabled' option.
> 
> Is there anything else I should check? Would it be worth going back to even older versions of GoldHEN? Not totally sure where I could be going wrong / how to even troubleshoot where the problem lies. Thanks!
> 
> -- Edit: I have also tried booting via the python script instead of the c++ (I don't think this would make a difference, but just something that I tried), no luck either way unfortunately.

Try using Systemx23x's fork at https://github.com/Systemx23x/PI-Pwn-AIO/tree/main/PPPwn

With this fork, internet passthrough works with goldhen 2.4b18, I think it's recent and good enough for me. It doesn't work with 2.4b18.2.

mistersoup3, 2025-01-20:
Finally got some free time to try this, but unfortunately same result. Downloaded 2.4b18, re-setup the PI with PPPwn from the Systemx23x fork, and am still unable to get the passthrough working as expected. Glad it's working for someone at least, but I think I'm about out of ideas for now, hoping a future update resolves it.

durag5, 2025-01-20:
I had DNS block enabled after a reinstall o forgot to disable. After
disabling that option in pi-pwn everything worked again. I'm back on 17 and
have not tried again on 18.

ShamelessWizard, 2025-01-21:
The stage2 binary is hard coded in the source. That needs to match the stage2 released in the goldhen release or else it will be problematic. I was going to make a patch and try it, but I took the path of laziness with the fork.

lufeig, 2025-01-22:
It seems that this fork has been updated to work with gh 2.4b18.2

https://github.com/Systemx23x/PI-Pwn-AIO/tree/main

ShamelessWizard, 2025-01-23:
> It seems that this fork has been updated to work with gh 2.4b18.2
> 
> https://github.com/Systemx23x/PI-Pwn-AIO/tree/main
> 
> EDIT: as of today, 22-jan-2025, internet passthrough is not working properly on this fork too.

I updated my local pppwn.py to include a byte encoded stage2 provided by Goldhen 2.4b18.2. 

PPPOE only worked once during all of my tests and that one time had a weird error. All other times it disconnects and never reconnects to the PPPOE service again.

Something is clearly off with 2.4b18.2.  2.4b18 works with the above release if you follow the installation instructions. I haven't restarted since it was working. Maybe it's another fluke.

adamvnz, 2025-01-29:
I've been having the same issue. Thought it was the way I set up the pi 4b. After several rebuilds and reseting the ps4 still no luck

lufeig, 2025-02-13:
so i have just tested the latest update from this fork and it's working with goldhen 2.4b18.2, including internet passthrough.

https://github.com/Systemx23x/PI-Pwn-AIO

ShamelessWizard, 2025-02-15:
I had luck with that after I used and forced it to use the 9.60 stage2 binary from goldhen. I haven't looked deeply at the stage2 code, but making the assumption that 950 is fine for 960 has a caused me a headache.

autumn-512, 2025-04-05:
Had this issue for months, can confirm that the solution for me was disabling the PPPoe Patch option in Goldhen settings. This setting disables the PS4 ethernet port after loading Goldhen to prevent the console from being exploited more than once. The setting is not necessary with Pi-Pwn and disabling it is such an obvious solution in hindsight.

ShamelessWizard, 2025-04-05:
> Had this issue for months, can confirm that the solution for me was disabling the PPPoe Patch option in Goldhen settings. This setting disables the PS4 ethernet port after loading Goldhen to prevent [...]

Ugh that’s not obvious at all! I’ve been running network dumps of the traffic and experimenting with why the client stops halfway through the pppoe handshake. I found a half dozen bugs at least.