stooged/PI-Pwn #211: Eating SD cards

durag5, 2025-02-02:
I only have raspberry pi lite 64 bit with Pi-pwn and Hyperion installed and it's rapidly killing the SD cards. Is their a lot of writing going on? The first card only had Hyperion and had worked for years. Also using same power supply, the "official" one from the cannakit. This is a rpi 3b+. I'm going to get another card to run on my rpi3 to rule that out.

stooged, 2025-02-02:
there are writes in the log files...
`/boot/firmware/PPPwn/pwn.log` if you enable verbose pppwn
`/boot/firmware/PPPwn/upd.log` when you run the update

when you save settings in the web ui `/boot/firmware/PPPwn/config.sh` and `/boot/firmware/PPPwn/pconfig.sh` are written to.

none of this is really major data or anything different to many different types of pi projects with log files and settings and other than those files everything is reading.

killing power during a write cycle may cause issues though because sd cards dont really like that.

i run hyperion on my 100" with rpi4 and have done for quite some time now and have never had an issue with that too.

durag5, 2025-02-02:
I have the power always on because I am also running Hyperion at all times.
So it shouldn't be a power cutting issue.