stooged/PI-Pwn #216: Debian 12 (Bookworm) – Missing /etc/rc.local (sed: can't read /etc/rc.local: No such file or directory)

mariozelaschi, 2025-03-18:
Hi all,

I'm using a Raspberry Pi 3 B+ with Raspberry Pi OS Lite (Debian 12 Bookworm), and the script fails with the following error:
`sed: can't read /etc/rc.local: No such file or directory
`
As I understand, Debian 12 has deprecated /etc/rc.local, which is no longer included by default.

To work around this, I manually recreated /etc/rc.local and configured it as a systemd service. However, this feels more like a workaround than a proper fix.

Would it make sense to update the script to support systemd natively instead of relying on /etc/rc.local? Alternatively, if maintaining compatibility is a priority, this workaround could at least be documented.

Or am I missing something? Any insights would be appreciated.

Thanks!

ShamelessWizard, 2025-03-19:
The sed commands are also very fragile. 

There was a bit of a hybrid period even with systemd and init scripts that had rc.local still being executed. It's probably just a hold-over.

To convert t

Should probably just re-implement devboot as a oneshot systemd startup script and then have pipwn service take a depdency on devboot. Then the sed changes to rc.local could just be changed to clean up artifacts.

```
[Unit]
Description=Run PPPwn devboot.sh once at startup
After=network.target

[Service]
Type=oneshot
ExecStart=/bin/bash /boot/firmware/PPPwn/devboot.sh
RemainAfterExit=yes
StandardOutput=journal
StandardError=journal

[Install]
WantedBy=multi-user.target
```

Then pipwn service will need to depend on devboot

```
[Unit]
Description=PiPwn Service
After=network.target pppwn-devboot.service
Requires=pppwn-devboot.service

[Service]
WorkingDirectory=/boot/firmware/PPPwn
ExecStart=/boot/firmware/PPPwn/run.sh
Restart=never
User=root
Group=root
Environment=NODE_ENV=production

[Install]
WantedBy=multi-user.target
```

Updated devboot script with some safety checks
```
#!/bin/bash -x

# Create base directory if it doesn’t exist
BASE_DIR="/media/pwndrives"
mkdir -p "$BASE_DIR" || { echo "Failed to create $BASE_DIR"; exit 1; }

# Get list of partitions, filtering and cleaning with lsblk
readarray -t usbarr < <(sudo lsblk -o NAME,TYPE | grep "part" | sed 's/[^[:alnum:]/ ]//g')

# Loop through each partition
for dev in "${usbarr[@]}"; do
    # Extract partition name (e.g., sda1) and ensure it’s valid
    PARTITION=$(echo "$dev" | awk '$2 == "part" {print $1}')
    
    # Skip if no valid partition is found
    if [ -z "$PARTITION" ]; then
        echo "Skipping invalid entry: $dev"
        continue
    fi

    # Set mountpoint path
    MOUNTPOINT="$BASE_DIR/$PARTITION"

    # Create mountpoint directory if it doesn’t exist
    mkdir -p "$MOUNTPOINT" || { echo "Failed to create $MOUNTPOINT"; continue; }

    # Mount the partition in the background if not already mounted
    if ! mountpoint -q "$MOUNTPOINT"; then
        sudo mount "/dev/$PARTITION" "$MOUNTPOINT" & 
        if [ $? -ne 0 ]; then
            echo "Failed to mount /dev/$PARTITION to $MOUNTPOINT"
        else
            echo "Mounted /dev/$PARTITION to $MOUNTPOINT"
        fi
    else
        echo "$MOUNTPOINT is already mounted"
    fi
done
```

ShamelessWizard, 2025-03-20:
A little logic to clean up the old devboot rc.local if it exists. This could replace the current sed logic to modify rc.local.

RC_LOCAL="/etc/rc.local"
TARGET_LINE="bash /boot/firmware/PPPwn/devboot.sh"

if [ -f "$RC_LOCAL" ] && grep -Fx "$TARGET_LINE" "$RC_LOCAL" > /dev/null; then
    sed -i "/^$TARGET_LINE$/d" "$RC_LOCAL"
    echo "Line removed from $RC_LOCAL"
fi