stooged/PI-Pwn #219: Error in the end

Rakanitshu, 2025-04-16:
Hi everyone, im trying install on my rasp pi 2 model b the pppwn and for 2 times give me 3 errors, dont understand what im missing?

 Failed to fetch http://raspbian.raspberrypi.com/raspbian/pool/main/g/git/git-                                                                                                man_2.39.5-0%2bdeb12u1_all.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 8                                                                                                0]
E: Failed to fetch http://raspbian.raspberrypi.com/raspbian/pool/main/g/git/git_                                                                                                2.39.5-0%2bdeb12u1_armhf.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?

ShamelessWizard, 2025-04-16:
What was the output of sudo apt-get update?

Rakanitshu, 2025-04-17:
Hi Shameless, i solve it, today do line by line on commands to install pppwn and work it, it tries multiples times to pppwn but maybe my rasp 2 model b its too slow to fast process it, buts its working ty.

Rakanitshu, 2025-04-17:
Btw, after put it in rest mode i realize theres a blinking white light on ps4, it doesnt start pressing power button or in ps command, need to force shut down and realize a fresh pppwn on it, any solution for it?

ShamelessWizard, 2025-04-17:
Make sure Goldhen has rest mode support enabled. Do not leave mini-apps open as they can cause a crash when in rest mode. It's recommended to close games as well. It's just a roll of the dice after that. I've had rest mode function for nearly two weeks even with a game open. Mini-apps are a known issue.

Rakanitshu, 2025-04-17:
Ok, the rest mode on goldhen for sure are active, but maybe I left some app from hbshop open, ty Mr. Wizard for all the help!