stooged/PI-Pwn #221: Can't disable/modify default DNS conf?

Parotaku, 2025-04-23:
Hello

I'm on 11.00 and everything is working perfectly in loading Goldhen with my PI3B+.

The only problem I got is I can't access my retail games updates in itemzflow: I can detect them but cannot download them because I get a message telling the servers are blocked by the DNS conf used in the PI.

Maybe I'm doing something wrong but I tried setting 'DDNS=true' in the config.sh file in the PPPwn folder, but that didn't change anything... doesn't DDNS means 'Disable DNS'?
Any help would be nice. Thanks.

ShamelessWizard, 2025-04-23:
If you select disable dns blocker in the web ui it will unset the values in  /etc/dnsmasq.more.conf which block playstation dns records. 

If you do this your playstation can contact Sony servers.

Parotaku, 2025-04-23:
Sorry if I sound like a total noob in this, but how do I access this web ui? (I have only configured my PI using the command line console)

ShamelessWizard, 2025-04-23:
I assume you know the IP address of your Pi. Enter that IP into a web browser. like http://192.168.69.69. Your ip address will vary based on your network configuration.

Parotaku, 2025-04-23:
Enabling, then disabling it again using the web ui checkbox did the trick.
Thanks for your help, ShamelessWizard.