stooged/PI-Pwn #224: Error sending packet: send: No such device or address - Pi Zero W USB to Ethernet

mult1v4c, 2025-06-10:
I followed all installation steps but I seem to be getting the same error over multiple tries using Pi Zero W with USB to Ethernet Adapter. It repeats almost at the exact same time. Already tried setting the buffer size to 10240 but still met with the same error. PI-Pwn ran between 30 minutes to two hours (I forgot about it and left it running). Using python doesn't seem to generate any logs for me. See more details below:

Pi Model = `Pi Zero W Rev 1.1`
OS = `Pi OS Lite Bookworm (Linux 6.12.25+rpt-rpi-v6 armv6l)`
PS4 Model = `CUH-2216A`
PS4 FW = `11.00`

Error snippet: [(Full Logs)](https://gist.github.com/mult1v4c/b7f78a1f8ad054856817b53353cd5786)

```
[ERROR: /build/_deps/pcapplusplus-src/Pcap++/src/PcapLiveDevice.cpp: sendPacket:628] Error sending packet: send: No such device or address
```

`config.sh`:
```sh
#!/bin/bash
INTERFACE="eth0"
FIRMWAREVERSION="11.00"
SHUTDOWN=false
USBETHERNET=true
PPPOECONN=true
VMUSB=false
DTLINK=false
PPDBG=true
TIMEOUT="5m"
RESTMODE=true
PYPWN=false
LEDACT="status"
DDNS=false
OIPV=false
UGH=true
```

`pconfig.sh`:
```sh
#!/bin/bash
XFWAP="1"
XFGD="4"
XFBS="0"
XFSN="0x1000"
XFPN="0x1000"
XFCN="0x1"
XFNWB=false
```
Please let me know if I can provide more information. Thanks!

mult1v4c, 2025-06-10:
```bash
$ ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether 00:e0:4c:53:44:58 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DORMANT group default qlen 1000
    link/ether b8:27:eb:9e:ba:3c brd ff:ff:ff:ff:ff:ff
```
```bash
$ ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 00:e0:4c:53:44:58 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::50f5:74d4:aa08:60c8/64 scope link noprefixroute
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:9e:ba:3c brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.91/24 brd 192.168.100.255 scope global dynamic noprefixroute wlan0
       valid_lft 86319sec preferred_lft 86319sec
    inet6 fe80::2488:8234:bab:7d4f/64 scope link noprefixroute
       valid_lft forever preferred_lft forever
```
```bash
$ dmesg | grep eth
[   29.163065] dm9601 1-1:1.0 eth0: register 'dm9601' at usb-20980000.usb-1, Davicom DM96xx USB 10/100 Ethernet, 00:e0:4c:53:44:58
[   60.230821] dm9601 1-1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xFFFF
```
```bash
$ lspci
pcilib: Cannot open /proc/bus/pci
lspci: Cannot find any working access method.
```
```bash
$ lsusb
Bus 001 Device 002: ID 0fe6:9700 ICS Advent DM9601 Fast Ethernet Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

ShamelessWizard, 2025-06-11:
Is /proc mounted? 

mount|grep proc

mult1v4c, 2025-06-11:
I checked as you said and /proc seems to be mounted:
```bash
$ mount | grep proc
proc on /proc type proc (rw,relatime)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=30,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
binfmt_misc on /proc/sys/fs/binfmt_misc type binfmt_misc (rw,nosuid,nodev,noexec,relatime)
```
fstab looks ok to me:
```bash
$ cat /etc/fstab
proc            /proc           proc    defaults          0       0
PARTUUID=0332d286-01  /boot/firmware  vfat    defaults          0       2
PARTUUID=0332d286-02  /               ext4    defaults,noatime  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
```
I mounted /proc anyway `sudo mount -t proc proc /proc` and restarted the PS4 with the Pi (powered via usb port) and it pwned it the first try (really thought you're an actual wizard lol)

I tried reproducing it if it still pwns it but the same error `No such device or address` persists.

Here's the ethtool info as requested:
```bash
Settings for eth0:
        Supported ports: [ TP    MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
        Link partner advertised pause frame use: Symmetric Receive-only
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 100Mb/s
        Duplex: Full
        Auto-negotiation: on
        Port: MII
        PHYAD: 0
        Transceiver: internal
        Current message level: 0x00000007 (7)
                               drv probe link
        Link detected: yes
```

mult1v4c, 2025-06-11:
I figured after my multiple attempts it may be required to have a certain level of luck when running the exploit haha. 

Anyways, thank you for your efforts Mr. Wizard! I am grateful for your time and assistance in troubleshooting this otherwise unsolvable issue. I'll take your advise and stick with this for now. Cheers!