stooged/PI-Pwn #225: Update to the latest GoldHen changes

narcyzzo, 2025-08-09:
https://github.com/SiSTR0/PPPwn/commit/164004894a674f2b065653a9d075f1c8efe7f24e
@stooged

suckingalemon, 2025-11-11:
> Hello! Since the repository hasn't been updated for a while, does anyone know how to manually update to the latest version of GoldHen and stage2? Thanks <3

Hey. You can download the latest version of GoldHEN (GoldHEN v2.4b18.6 at the time of writing) from [here](https://ko-fi.com/sistro/shop). Add the `.bin` file to the root of USB stick. It will be transferred to your PS4 when you next run `PI-Pwn`. As for updating `stage2`, I am unsure.

narcyzzo, 2025-11-11:
stage 2 files are here, just paste new files here :) https://github.com/stooged/PI-Pwn/tree/main/PPPwn/gh_sta
local files path /boot/firmware/PPPwn
if updating goldhen bin just make a pendrive with updated bin and run the exploit once with it, new updated goldhen will be copied

mariozelaschi, 2025-11-13:
In case you are interested, I forked the repo and made some updates (compatibility with Debian 13, GoldHEN updated, etc.): https://github.com/mariozelaschi/PI-Pwn

mariozelaschi, 2025-11-14:
This release didn't have any update from some time now, so if you want to use it on a current Debian 13 and you want the newest GoldHEN you need to do some manual adjustments.
I will update my fork today with instructions and updated setup to take into account who already has the old version installed.

HunterAhlquist, 2025-11-15:
> In case you are interested, I forked the repo and made some updates (compatibility with Debian 13, GoldHEN updated, etc.): https://github.com/mariozelaschi/PI-Pwn

Hi, you don't have an issues page so I'd thought I tell you here. 

I tried your fork but on the one-liner I get: 
```
This script requires root privileges. Restarting with sudo...
/usr/bin/bash: /usr/bin/bash: cannot execute binary file
```
and when trying the interactive setup or manual I get
```
install.sh: line 699: syntax error near unexpected token `else'
install.sh: line 699: `else'
```
after picking the setting for verbose mode. I tried these through macOS' terminal SSH client to my Pi 4b and with Raspberry Pi OS Lite 64bit

this is the config I picked:
```
Do you want to enable the option to use Python (slower) PPPwn (Y|N)?: n
The Python version of PPPwn will not be available
dpkg-query: no packages found matching vsftpd

Do you want to install a FTP server? (Y|N): n
Skipping FTP installation
dpkg-query: no packages found matching samba

Do you want to setup a SAMBA share? (Y|N): n
Skipping Samba installation
dpkg-query: no packages found matching python3-scapy

Do you want to change the PPPoE username and password?
If you select No, these defaults will be used:

Username: ppp
Password: ppp

(Y|N)?: n
Using default settings:

Username: ppp
Password: ppp

"ppp"  *  "ppp"  192.168.2.2

Do you want to detect console shutdown and restart PPPwn?

(Y|N)?: y
Console shutdown detection enabled

Do you want the console to connect to the internet after PPPwn? (Y|N): y
Console internet access enabled

Are you using a USB to Ethernet adapter for the console connection?

(Y|N)?: n
USB to Ethernet adapter will NOT be used

Do you want to detect if GoldHEN is already running and skip PPPwn if found?
(Useful for rest mode)

(Y|N)?: y
GoldHEN detection enabled

Do you want PPPwn to run in verbose mode?

(Y|N)?: n
PPPwn will NOT run in verbose mode
```

mariozelaschi, 2025-11-16:
@HunterAhlquist thanks for the message! 
I have added the issues page to my repo.
I am working at the bugs, the updated versions should be online by tonight